theori-io/nrsc5 #401: Decode HERE images

argilo, 2025-05-03:
#308 made it possible to receive packet & stream data through the API, but leaves decoding the data as an exercise to the reader. The ensuing discussion thread has lots of details about the contents of  the "HERE images" stream. It may be useful for nrsc5 itself to handle decoding of this stream, making the data available through a new API interface.

closed 2025-05-31 by argilo

markjfine, 2025-06-04:
That would be great! Thanks.

markjfine, 2025-06-04:
Noticed something else that indicates a data error: The traffic image is square, with 0.64306 delta degrees in height, 0.82397 delta degrees in width. The weather image is not square, with 0.824 delta degrees in height, and 0.82399 delta degrees in width. Someone clearly thought that to make a square, the height in degrees should equal the width, not understanding how projections work. Should also point out that although the actual lat/lons are shifted a little, the delta widths for traffic and weather are very close, whereas the delta heights for weather are 0.18094 degrees taller - quite a bit at this scale. Again, this seemingly points to an error in the upper left lat for weather.

argilo, 2025-06-04:
Images: https://csclub.uwaterloo.ca/~cd3smith/nrsc5/here.zip

Let me know if there's anything more I can get you from those recordings.

markjfine, 2025-06-04:
> Let me know if there's anything more I can get you from those recordings.

@argilo 
Thanks for those. Would it be possible to get the IQ recording for WOMC (if that's what was used to create the images)?

markjfine, 2025-06-04:
Cheers.

argilo, 2025-06-04:
https://csclub.uwaterloo.ca/~cd3smith/nrsc5/detroit-1043-HERE.cu8

argilo, 2025-06-04:
If you want to double check the md5 sum, it's `535bdfee209aa706568dab402825ab69`.

markjfine, 2025-06-04:
Yup.. we're good! Thanks again.

argilo, 2025-06-05:
There's something odd about the traffic coordinates. The coordinates for the `trafficMap_1_1` tile (marked "5" and "14" in the image below) seem to be valid for the traffic image as a whole. The lat1/lon1 coordinates (marked in red) seem to move in the opposite direction from expected, and the lat2/lon2 coordinates (marked in blue) move in the correct direction, but are offset from where they should be.

I wonder if there could be something wrong with the coordinate parsing.

[image]

markjfine, 2025-06-05:
Likely a sign error or a variable not passing somewhere.
Also, you would think that given this scheme, 5 and 10, 6 and 11, 8 and 13, and 9 and 14 would each be colocated. Didn't notice that before - was too focused on the recurrence of 0.824 in the weather. Didn't really pay attention to the coordinates for travel, since the blocks seam together without it.

markjfine, 2025-06-06:
Just had an weird thought: What if those weren't 26-bit integers that have to be divided by 100000, but retro Borland-style 24-bit reals (if that's not dating myself).

markjfine, 2025-06-06:
Something else that seems at play: Because of the weird weather coordinate thing, I started creating the weather backdrop using the upper left coordinates for traffic tile 0,0 and the lower right coordinates for tile 2,2. As you can see, they're not that far off, so maybe something else is going on?

[image]

[image]

markjfine, 2025-06-06:
Here is the backdrop I created using the WMOC iq file, that correlates to your map above:

[image]

markjfine, 2025-06-06:
I played a bit with here_images.c, to include using long long instead of int to see if something might be getting crushed. But that's not the case. I'm reticent to try to map all this to a bit structure and read it directly into variables that way, because although it would be an interesting thing to do, I don't think it'd do much good.

Segue to... Trying to think about this problem logically (because this is driving my OCDs bonkers):
1. Let's assume lat1 and lon1 of tile 0,0 is in fact correct, and that lat2 and lon2 of tile 2,2 is correct, which I think we can agree seems to be the case.
2. If that's true, it means the bit-wise calculations for each of the files are actually correct, since the calculations don't care which of the tiles is being used, all things being equal.
3. It also means that HERE is likely populating everything else in-between (and their weather overlay) with incorrectly calculated garbage. That would suck considering they portend to be a 'nav-tech' company (sorry for the pun).
4. I would also tend to think that they have pre-defined images with lookup tables for their weather overlays that are tied to the location ID (e.g., c37e for Detroit or 6hvu for DC). In this case, the actual coordinates would be irrelevant to their system so adding them may just follow an arbitrary proforma spec (that they themselves created). Unfortunately, that's not the case for us outsiders.

You learn to get cynical about such things when you're 66.

Wonder if anyone's actually seen what the Navteq/HERE data looks like.

argilo, 2025-06-06:
It's certainly possible that receivers ignore the lat/lon data fields. Sending modified data to a receiver would be an interesting experiment, but I expect that won't work until the 16-bit CRC / checksum is reverse engineered.

I would guess that receivers are only available in cars. This blog post mentions "certain 2017 [Hyundai] vehicle models sold in the US and Canada."

https://www.here.com/learn/blog/here-hd-traffic-offers-more-traffic-updates-with-lower-data-costs

markjfine, 2025-06-06:
Yeah... I saw that and a few others, but nothing recent regarding them and HD Radio. The map they show in that article looks almost too detailed for a car.

PR #415 (2025-05-31): Decode HERE Images
Fixes #401.

Most of the details of the HERE Images stream were worked out in #308, so I've now added a decoder for it. Images are reported in `NRSC5_EVENT_HERE_IMAGE` API events, and logged by the C and Python command-line applications:

```
15:56:09 HERE Image: type=TRAFFIC, seq=2, n1=1, n2=9, time=2024-10-23T13:02:12Z, lat1=43.02372, lon1=-83.81881, lat2=42.82260, lon2=-83.54415, name=trafficMap_0_0_rdhs.png, size=3047
15:56:09 HERE Image: type=TRAFFIC, seq=2, n1=2, n2=9, time=2024-10-23T13:02:12Z, lat1=43.02372, lon1=-84.09347, lat2=42.82260, lon2=-83.26950, name=traffic [... cut]