theori-io/nrsc5 #405: Callback Sis Event Has Too Many Fields

TheDaChicken, 2025-05-07:
While #402 was being worked on, @argilo and I had realized this. As new features were added to NRSC5 over the past years, the SIS event also grew. There are now 18 fields. 

Since a ton of fields are included in the single sis event, this requires the API callback to constantly process known information. 

If one of these 18 fields changes, the client is required to process each field in the whole sis event again. For fields such as data services, audio services, station name which rarely change don't need to be processed again. 

In the nrsc5 ctl application, we see this as console spam where information is repeated a few times. Such as Station Name, Data Services, Audio Services.

Events such as ID3 can be argued to be repeatable because ID3 contains data that changes often. 

There is a huge problem: ***improving sig event requires making modifications to the ABI design.***

Here's a start with some ideas:
- Keep the old sis event while making the event "deprecated." Then, create new event types for emergency etc. After "pretty long time," the old sig event can be removed.

argilo, 2025-05-07:
Agreed that the SIS event is getting a bit large. It at least won't get any larger, since we're already decoding and reporting all SIS message types.

Splitting into several smaller SIS events sounds like a good idea. But I'd say the change is a low priority, since the current API does work, even if it's clumsy.

markjfine, 2025-05-07:
Could be even more: If you count four for device and synchronization detection and potentially two more for Navteq/HERE (stream/packet data messages that are holdovers from when we were testing options to decode this a while back), -dui currently has 25 regex strings to detect useful messages.

argilo, 2025-05-07:
I'd propose the following breakdown:

* Station ID
  * country_code
  * fcc_facility_id
* Station Name
  * name
* Station Slogan
  * slogan
* Station Message
  * message
* Station Location
  * latitude
  * longitude
  * altitude
* Audio Service
  * program
  * access
  * type
  * sound_exp
* Data Service
  * access
  * type
  * mime_type
* Emergency Alert
  * alert
  * alert_cnt
  * alert_cnt_length
  * alert_category1
  * alert_category2
  * alert_location_format
  * alert_num_locations
  * alert_locations

Audio Service and Data Service descriptors could be sent one at a time instead of in a linked list.

How does this sound?

argilo, 2025-05-07:
The breakdown above is almost a one-to-one mapping to SIS message types, with two exceptions:

* "Service Information Message" is split into Audio Service and Data Service, since they have different formats.
* Station Name is merged from "Station Name (short format)", "Station Name (long format)", and "Universal Short Station Name", which are three different versions of the same thing from different versions of the NRSC-5 standard.

argilo, 2025-05-07:
> It at least won't get any larger

While reviewing the SIS document, I realized that there's actually more to come. We decode SIS Parameter Messages (type 7), but don't yet expose them through the API:

https://github.com/theori-io/nrsc5/blob/a0f5894c0376a8c90850f8d34d86727c7c9e6dc3/src/pids.c#L600-L663

It would be much easier to add these to the API if they could go in separate event types.

TheDaChicken, 2025-05-08:
> I'd propose the following breakdown:
> 
> * Station ID
>   * country_code
>   * fcc_facility_id
> * Station Name
>   * name
> * Station Slogan
>   * slogan
> * Station Message
>   * message
> * Station Location
>   * latitude
>   * longitude
>   * altitude
> * Audio Service
>   * program
>   * access
>   * type
>   * sound_exp
> * Data Service
>   * access
>   * type
>   * mime_type
> * Emergency Alert
>   * alert
>   * alert_cnt
>   * alert_cnt_length
>   * alert_category1
>   * alert_category2
>   * alert_location_format
>   * alert_num_locations
>   * alert_locations
> 
> Audio Service and Data Service descriptors could be sent one at a time instead of in a linked list (similar to #402).
> 
> How does this sound?
> 

I think this sounds good. As silly as this is for making many new events, this does make things a lot easier. Especially since not everything in the sis is exposed to the API.

argilo, 2025-05-08:
Yeah, even though it's a lot of event types, I think it will ultimately make life easier both for libnrsc5 (which won't need to use an `updated` flag to keep track of whether any SIS fields have changed) and API clients (which won't have to deal with redundant updates).

argilo, 2025-05-08:
Here's a prototype version, which thus far splits Station ID and Station Name into their own event types:

https://github.com/theori-io/nrsc5/compare/split-sis-api

The reduction in log traffic at startup is already noticeable.

Any early feedback on the API change would be appreciated.

argilo, 2025-05-08:
The only thing that stands out for me is that the names (in C) are getting longer:

Old:
`evt->sis.country_code`
`evt->sis.fcc_facility_id`
`evt->sis.name`

New:
`evt->sis_station_id.country_code`
`evt->sis_station_id.fcc_facility_id`
`evt->sis_station_name.name`

argilo, 2025-05-08:
Perhaps we could delete `SIS_` / `sis_` from the event & struct names?

TheDaChicken, 2025-05-08:
Removing SIS_ from the event & struct name could work. There is a problem when we hit Audio Service. `evt->audio_service` exists a name already. 

We could shorten the names of the fields like `country_code` to `cc`. This may make things unreadable. 

This may not be exactly much of an issue if these structs had a typedef definition. It can be possibly shorten in code like so:

```c
nrsc5_sis_station_id* station_id = &evt->sis_station_id;

station_id.country_code
station_id.fcc_facility_id
```

argilo, 2025-05-08:
Slogan, message, and location are now added. And I removed the `SIS_` prefixes.

PR #409 (2025-05-08): Split up the SIS event
Fixes #405.

Here I've deprecated the SIS event and broken it down into eight smaller events:

* Station ID
* Station Name
* Station Slogan
* Station Message
* Station Location
* Audio Service Descriptor
* Data Service Descriptor
* Emergency Alert

Doing so has already helped remove a lot of redundant log messages from the C and Python CLI applications.

Before:
```
14:05:03 Synchronized
14:05:03 Country: CA, FCC facility ID: 328
14:05:03 Station name: MOVE-FM
14:05:03 Country: CA, FCC facility ID: 328
14:05:03 Station name: MOVE-FM
14:05:03 Audio program 0: public, type [... cut]
closed 2025-05-08 by argilo