theori-io/nrsc5 #423: Python ctl cannot find libnrsc5.dll

TheDaChicken, 2025-06-02:
Currently, the Python ctl uses `ctypes.cdll.LoadLibrary(lib_name)` with a relative path assuming that the function will find the library in path.

Sadly, this behavior in Windows doesn't work. In Python 3.8, a new parameter called winmode was added to `ctypes.CDLL(lib_name)`. 

Some context: the `ctypes.cdll.LoadLibrary(lib_name)` function is a wrapper around `ctypes.CDLL(lib_name)`. The function constructs the class with its default arguments. The default argument for the winmode parameter is None causing the function to only search the current directory for libraries on Windows. 

We can set the winmode parameter to 0 to bring back this behavior. Though checking the Python documentation on [ctypes.CDLL](https://docs.python.org/3/library/ctypes.html#ctypes.CDLL) for winmode, Python prefers that the exact path to the library is provided.

argilo, 2025-06-02:
When I built the Python API wrapper and CLI, I would have tested with Python 3.7 which was current at the time. It does look like things changed in Python 3.8:

https://docs.python.org/3/whatsnew/3.8.html#ctypes

I'll have to have a closer look and see whether it would be appropriate to override the new default behaviour.

argilo, 2025-06-02:
More details here: https://docs.python.org/3/whatsnew/3.8.html#bpo-36085-whatsnew

> DLL dependencies for extension modules and DLLs loaded with [`ctypes`](https://docs.python.org/3/library/ctypes.html#module-ctypes) on Windows are now resolved more securely. Only the system paths, the directory containing the DLL or PYD file, and directories added with [`add_dll_directory()`](https://docs.python.org/3/library/os.html#os.add_dll_directory) are searched for load-time dependencies. Specifically, **PATH and the current working directory are no longer used**, and modifications to these will no longer have any effect on normal DLL resolution. If your application relies on these mechanisms, you should check for [`add_dll_directory()`](https://docs.python.org/3/library/os.html#os.add_dll_directory) and if it exists, use it to add your DLLs directory while loading your library. Note that Windows 7 users will need to ensure that Windows Update KB2533623 has been installed (this is also verified by the installer). (Contributed by Steve Dower in [bpo-36085](https://bugs.python.org/issue?@action=redirect&bpo=36085).)

argilo, 2025-06-02:
This would allow the DLL to be loaded from the same directory as `nrsc5.py`:

```diff
diff --git a/support/nrsc5.py b/support/nrsc5.py
index 7174277..46086fe 100644
--- a/support/nrsc5.py
+++ b/support/nrsc5.py
@@ -3,6 +3,7 @@ import ctypes
 import datetime
 import enum
 import math
+import os
 import platform
 import socket
 
@@ -581,6 +582,7 @@ class NRSC5:
     def _load_library(self):
         if NRSC5.libnrsc5 is None:
             if platform.system() == "Windows":
+                os.add_dll_directory(os.path.dirname(__file__))
                 lib_name = "libnrsc5.dll"
             elif platform.system() == "Linux":
                 lib_name = "libnrsc5.so"
```

Do you think that's a reasonable approach? Or do you have a different proposal about where we should allow the DLL to be loaded from?

argilo, 2025-06-03:
@TheDaChicken suggested this as another option:

```diff
diff --git a/support/nrsc5.py b/support/nrsc5.py
index 7174277..4b0e99e 100644
--- a/support/nrsc5.py
+++ b/support/nrsc5.py
@@ -3,6 +3,7 @@ import ctypes
 import datetime
 import enum
 import math
+import os
 import platform
 import socket
 
@@ -581,7 +582,7 @@ class NRSC5:
     def _load_library(self):
         if NRSC5.libnrsc5 is None:
             if platform.system() == "Windows":
-                lib_name = "libnrsc5.dll"
+                lib_name = os.path.join(os.path.dirname(__file__), "libnrsc5.dll")
             elif platform.system() == "Linux":
                 lib_name = "libnrsc5.so"
             elif platform.system() == "Darwin":
```

This would load the DLL only from the directory where `nrscy.py` resides. One downside I can see is that any existing install would have to have put libnrsc5.dll in the system path, and that would no longer work.

PR #426 (2025-06-03): Allow loading libnrsc5.dll from the same directory as nrsc5.py on Windows
Fixes #423.

For security reasons, Python versions 3.8 and later no longer load DLL files from the PATH or the current directory:

https://docs.python.org/3/whatsnew/3.8.html#bpo-36085-whatsnew

At present, the only remaining option would be to install libnrsc5.dll to the system path. It would be convenient to allow the library to be loaded from the same directory as nrsc5.py, and this shouldn't increase the security risk unnecessarily.