theori-io/nrsc5 #424: Segmentation Fault on Windows when using nrsc5_open* on a linked Shared Build

TheDaChicken, 2025-06-02:
I discovered this issue a while ago in #323 

When I link to a shared build of libnrsc5 instead of a static build to use the api in a separate project, any related nrsc5_open* functions create a segment fault.

Using GDB shows that it happens on nrsc5_alloc:
[image]

TheDaChicken, 2025-06-02:
Something might be serious wrong somewhere. When linked to a shared build, the compiled program ends up looking for nrsc5.exe instead of nrsc5.dll. That doesn't seem right.

argilo, 2025-06-03:
I believe this is what's happening:

* On Windows, if a target "foo" has any exported symbols, then an "import library" named "libfoo.dll.a" is generated.
* The target name for our shared library is "nrsc5", and so its import library is named "libnrsc5.dll.a".
* The "app" executable target has its `OUTPUT_NAME` set to "nrsc5", and this target exports symbols (`nrsc5_*` and `NeAACDec*`). Therefore, it also generates an import library which named "libnrsc5.dll.a", overwriting the one for the shared library.

@TheDaChicken As a temporary workaround, you could set the `OUTPUT_NAME` of the "app" target to anything other than "nrsc5".

The exports coming from our "app" target seem to be a consequence of it being statically linked, and therefore inheriting the exports of the libraries it pulls in. So we might need to ensure that `libnrsc5.a` and `libfaad_hdc.a` do not have any exports.

argilo, 2025-06-03:
I had a peek at what librtlsdr does, and it specifically avoids using `__declspec(dllexport)` when building the static version of its library. But libnrsc5 and libfaad use it across the board. :-(

PR #425 (2025-06-03): Avoid exporting from static libraries & executables
Fixes #424.

To avoid accidentally exporting symbols, I've done two things here:

* build libnrsc5's shared and static versions separately, and only set `NRSC5_EXPORTS` for the shared version
* remove `__declspec(dllexport)` from faad2 in our patch; we're only building static libraries so it's not needed

With these changes, nrsc5.exe no longer has the spurious exports which were causing libnrsc5.dll.a to be overwritten.

I know very little about CMake & software builds, so I'd appreciate feedback on the approach.
closed 2025-06-03 by argilo