theori-io/nrsc5 #431: Dynamically update lines instead of printing new ones?

pclov3r, 2025-06-04:
This might require a bit of work but throwing the idea out there of dynamically update lines instead of printing new ones?

For example BER, MER, ID3 tags, etc. get updated in place instead of printing a new line. 

Perhaps this will need to be a optional feature however.

argilo, 2025-06-04:
I have thought about this too. At present, the CLI applications print out every API event as it arrives, and the flood of output is getting difficult for humans to consume.

The difficulty I would anticipate is in dealing with differences between terminals across different platforms. Ncurses would be a great solution (and it's even available in Python) except that it only supports POSIX operating systems (no Windows!).

Does anyone know whether it is possible to do this in a cross-platform (and not overly complicated) way?

argilo, 2025-06-04:
Msys2 does have a Windows ncurses package: https://packages.msys2.org/packages/mingw-w64-x86_64-ncurses

Python's built-in ncurses wrapper does not support Windows, sadly.

1000TurquoisePogs, 2025-10-23:
So, I wrote a little NodeJS program for my own sake on that idea. Maybe I'll learn how to do it in the python code later and submit a PR, but just putting this here to share

```javascript
const { spawn } = require('node:child_process');

const STRINGS_TO_MATCH = [
  'Station Name:',
  'Country:',
  'Station location:',
  'Slogan:',
  'Title:',
  'Artist:',
  'Album:',
  'Audio bit rate:',
  'BER:',
  'MER:',
  'XHDR:',
  'Audio service 0:',
  'Audio service 1:',
  'Audio service 2:',
  'Audio service 3:',
];

function updateLine(lineNumber, text) {
  console.log('\x1b[' + lineNumber + ';0H\x1b[0K' + text);
}

const nrsc5 = spawn('nrsc5', process.argv.slice(2));
nrsc5.stderr.setEncoding('utf8');

console.log('\x1b[0;0H\x1b[0J');

let dataBuffer = '';

nrsc5.stderr.on('data', (data) => {
  if (!data.includes('\n')) {
    dataBuffer += data;
    return;
  } else {
    let firstPart = data.substring(0, data.indexOf('\n'));
    let lastPart = data.substring(data.indexOf('\n'));
    data = dataBuffer + firstPart;
    dataBuffer = '';
  }

  for (let i = 0; i < STRINGS_TO_MATCH.length; i++) {
    let index = data.indexOf(STRINGS_TO_MATCH[i]);
    if (index != -1) {
      updateLine(i+1, data.substring(index));
      break;
    }
  }
});

nrsc5.on('close', (code) => {
  console.log(`nrsc5 closed, rc=${code}`);
});
```

https://github.com/user-attachments/assets/bcada041-8052-4d52-83c4-98ef556d36a3