theori-io/nrsc5 #433: Package Configuration Files for libnrsc5

TheDaChicken, 2025-06-04:
For my project that uses libnrsc5, I had to write my own cmake module to be able to find nrsc5: https://github.com/TheDaChicken/nrsc5_gui/blob/master/cmake/FindNRSC5.cmake

I realized this project could benefit providing packages configuration files for pkg-config and/or cmake to find libnrsc5 when this project is built/installed. 

That would make it easier including libnrsc5 into projects especially the static version which requires extra libraries for linking.

argilo, 2025-06-04:
Do we just need to upstream your FindNRSC5.cmake?

pclov3r, 2026-05-10:
Just bumping this would be very useful to have a .pc file for nrsc5. It would make it much easier for cmake to find nrsc5 as a dep for other applications using cmake that attempt to locate nrsc5 as a dep.