theori-io/nrsc5 #451: Does NRSC5 EAS contain start time and end time?

TheDaChicken, 2025-09-04:
The implementation of NRSC5's EAS decoding has some unknown data that is skipped. I came across this video where it shows time of the EAS alert which may or may not be apart of the message. https://youtu.be/5ZpY9mv9Es8 

I wonder if somewhere in NRSC5 EAS contains the time and duration/end time of the EAS alert?? 
In America, SAME contains the start and duration time: https://en.wikipedia.org/wiki/Specific_Area_Message_Encoding#Header_format for America

argilo, 2025-09-18:
The corresponding transmit code is here: https://github.com/argilo/gr-nrsc5/blob/7bb202d594f345934d3fd5a4f99f546f376f9aff/apps/send_alert.py#L60-L68

I expect the "unknown" bit values I put in there were the same as what I saw in all the alerts I had received.

Some clues about the control data can be found in U.S. patent 8,138,915: https://patents.google.com/patent/US8138915B2/en

Things may of course differ between what was patented, and what was actually implemented.

It seems the RX_IDD_2206 document may describe the control data, but sadly it's not public.

argilo, 2025-09-24:
Yes, it's certainly possible that the bit lengths could be different than what's shown in the patent.

Unfortunately we have too little data to draw any conclusions at this point. More recordings would definitely be helpful. Also, I'd be interested to know whether any receivers use the unknown bits. My Sangean receivers only seem to care about the category and location codes.

> The group of unknown 9 bits starts with 10. That would be two

Actually one, since integers are stored with the least-significant bit first.