theori-io/nrsc5 #452: Windows Compiled With Errors - no dll or exe Created

gpank2249, 2025-09-13:
Everything was going fine until the very end...

gordy@WIN11-i9 MINGW64 ~
$ pacman -Su
:: Starting core system upgrade...
 there is nothing to do
:: Starting full system upgrade...
resolving dependencies...
looking for conflicting packages...

Packages (6) curl-8.16.0-1  libcurl-8.16.0-1  libffi-3.5.2-1  libnghttp2-1.67.0-1
             libp11-kit-0.25.7-1  p11-kit-0.25.7-1

Total Download Size:   2.00 MiB
Total Installed Size:  6.17 MiB
Net Upgrade Size:      0.21 MiB

:: Proceed with installation? [Y/n] y
:: Retrieving packages...
 libp11-kit-0.25.7-1-x86_64      238.4 KiB   235 KiB/s 00:01 [###############################] 100%
 libcurl-8.16.0-1-x86_64         362.2 KiB   320 KiB/s 00:01 [###############################] 100%
 p11-kit-0.25.7-1-x86_64         471.5 KiB   402 KiB/s 00:01 [###############################] 100%
 libffi-3.5.2-1-x86_64            30.9 KiB   131 KiB/s 00:00 [###############################] 100%
 libnghttp2-1.67.0-1-x86_64       64.3 KiB  47.3 KiB/s 00:01 [###############################] 100%
 curl-8.16.0-1-x86_64            877.1 KiB   271 KiB/s 00:03 [###############################] 100%
 Total (6/6)                    2044.4 KiB   594 KiB/s 00:03 [###############################] 100%
(6/6) checking keys in keyring                               [###############################] 100%
(6/6) checking package integrity                             [###############################] 100%
(6/6) loading package files                                  [###############################] 100%
(6/6) checking for file conflicts                            [###############################] 100%
(6/6) checking available disk space                          [###############################] 100%
:: Processing package changes...
(1/6) upgrading libnghttp2                                   [###############################] 100%
(2/6) upgrading libffi                                       [###############################] 100%
(3/6) upgrading libp11-kit                                   [###############################] 100%
(4/6) upgrading p11-kit                                      [###############################] 100%
(5/6) upgrading libcurl                                      [###############################] 100%
(6/6) upgrading curl                                         [###############################] 100%
:: Running post-transaction hooks...
(1/1) Updating the info directory file...

gordy@WIN11-i9 MINGW64 ~
$ pacman -S git
resolving dependencies...
looking for conflicting packages...

Packages (34) heimdal-7.8.0-5  libcbor-0.12.0-1  libfido2-1.16.0-1  openssh-10.0p1-2
              perl-Authen-SASL-2.1900-1  perl-Clone-0.47-2  perl-Convert-BinHex-1.125-2
              perl-Encode-Locale-1.05-2  perl-Error-0.17030-1  perl-File-Listing-6.16-1
              perl-HTML-Parser-3.83-2  perl-HTML-Tagset-3.24-1  perl-HTTP-Cookies-6.11-1
              perl-HTTP-Daemon-6.16-1  perl-HTTP-Date-6.06-1  perl-HTTP-Message-7.00-1
              perl-HTTP-Negotiate-6.01-3  perl-IO-HTML-1.004-2  perl-IO-Socket-SSL-2.094-1
              perl-IO-Stringy-2.113-2  perl-LWP-MediaTypes-6.04-2  perl-MIME-tools-5.515-1
              perl-MailTools-2.22-1  perl-Net-HTTP-6.23-1  perl-Net-SMTP-SSL-1.04-2
              perl-Net-SSLeay-1.94-3  perl-TermReadKey-2.38-7  perl-TimeDate-2.33-2
              perl-Try-Tiny-0.32-1  perl-URI-5.32-1  perl-WWW-RobotRules-6.02-3
              perl-http-cookiejar-0.014-1  perl-libwww-6.79-1  git-2.51.0-1

Total Download Size:    9.84 MiB
Total Installed Size:  52.69 MiB

:: Proceed with installation? [Y/n] y
:: Retrieving packages...
 perl-Net-SSLeay-1.94-3-x86_64   219.4 KiB   212 KiB/s 00:01 [###############################] 100%
 perl-MIME-tools-5.515-1-any     176.8 KiB   167 KiB/s 00:01 [###############################] 100%
 heimdal-7.8.0-5-x86_64          544.5 KiB   397 KiB/s 00:01 [###############################] 100%
 perl-libwww-6.79-1-any          125.2 KiB   406 KiB/s 00:00 [###############################] 100%
 perl-IO-Socket-SSL-2.094-1-any  162.7 KiB   426 KiB/s 00:00 [###############################] 100%
 perl-URI-5.32-1-any              93.7 KiB   330 KiB/s 00:00 [###############################] 100%
 git-2.51.0-1-x86_64               6.8 MiB  3.77 MiB/s 00:02 [###############################] 100%
[... 320 lines cut ...]
    make[1]: *** [CMakeFiles/cmTC_2c612.dir/build.make:81: CMakeFiles/cmTC_2c612.dir/testCCompiler.c.obj] Error 1
    make[1]: Leaving directory '/home/gordy/nrsc5/build/CMakeFiles/CMakeScratch/TryCompile-2fjtk3'
    make: *** [Makefile:134: cmTC_2c612/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:7 (project)

-- Configuring incomplete, errors occurred!

gordy@WIN11-i9 MINGW64 ~
$ ^C

gordy@WIN11-i9 MINGW64 ~
$ cd ~/nrsc5/support

gordy@WIN11-i9 MINGW64 ~/nrsc5/support
$ xz -d sample.xz

gordy@WIN11-i9 MINGW64 ~/nrsc5/support
$ nrsc5.exe -r sample 0
-bash: nrsc5.exe: command not found

gordy@WIN11-i9 MINGW64 ~/nrsc5/support
$

TheDaChicken, 2025-09-13:
From my understanding, I don't see anything unusual. gcc was installed via pacman etc. 

In ```C:/msys64/home/gordy/nrsc5/build/CMakeFiles``` , there should be a more detailed log file named "CMakeError.log" containing the output of cc.exe. What's in there? 

BTW, I recommend quoting any large amount of text on GitHub like terminal input: https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code This can make reading terminal input easier

markjfine, 2025-09-14:
Msys/pacman may do things differently, but if gcc was properly installed, I would think it would be trying to use it (e.g., gcc*.exe). For some reason cmake may be picking a non-compliant compiler already provided with Msys called cc.exe. If this is the case, the gcc compiler may have to be specified in a command switch during cmake.

gpank2249, 2025-09-14:
ChatGPT to the rescue!

The commands for compiling to Windows should be:

Building on Windows with MSYS2
Install MSYS2. Open a terminal using the "MSYS2 MinGW 64-bit" shortcut. (Or use the 32-bit shortcut if you prefer a 32-bit build.)

pacman -Syu
If this is the first time running pacman, you will be told to close the terminal window. After doing so, reopen using the same shortcut as before.

pacman -Su
pacman -S git
git clone https://github.com/theori-io/nrsc5.git
**cd nrsc5  **this is what was missing**
nrsc5/support/msys2-build -j4

argilo, 2025-09-14:
That doesn't look correct. The nrsc5 directory is included in the final command, so changing into that directory is not necessary, and would cause that command to fail.